MnlPhlp/tauri-plugin-blec #5: Plugin init crashes app

aloulouamine, 2025-01-26:
Hello

 when launching an app with `tauri_plugin_blec::init()` it crashes at startup with : 

`Message from debugger: Terminated due to signal 9`

Target : Physical iphone 16 pro. 

`[✔] Environment
    - OS: Mac OS 15.2.0 arm64 (X64)
    ✔ Xcode Command Line Tools: installed
    ✔ rustc: 1.84.0 (9fc6b4312 2025-01-07)
    ✔ cargo: 1.84.0 (66221abde 2024-11-19)
    ✔ rustup: 1.27.1 (54dd3d00f 2024-04-24)
    ✔ Rust toolchain: stable-aarch64-apple-darwin (default)
    - node: 20.9.0
    - pnpm: 9.15.4
    - yarn: 4.0.0-rc.39
    - npm: 10.1.0
    - bun: 1.2.0

[-] Packages
    - tauri 🦀: 2.2.5
    - tauri-build 🦀: 2.0.5
    - wry 🦀: 0.48.1
    - tao 🦀: 0.31.1
    - @tauri-apps/api : 2.2.0
    - @tauri-apps/cli : 2.2.7

[-] Plugins
    - tauri-plugin-opener 🦀: 2.2.5
    - @tauri-apps/plugin-opener : 2.2.5

[-] App
    - build-type: bundle
    - CSP: unset
    - frontendDist: ../dist
    - devUrl: http://localhost:1420/
    - bundler: Vite

[-] iOS
    - Developer Teams: xxx (ID: xxxx)`

MnlPhlp, 2025-01-28:
do you get any other output where exactly the crash happened?
And does this happen with your own app or the example in this repo?

aloulouamine, 2025-01-28:
Example is working. I'm using it as starter. Thank you !

closed 2025-01-28 by aloulouamine