MnlPhlp/tauri-plugin-blec #6: How do you remove desktop platform?

tinyvan, 2025-02-12:
I'm developing a Tauri plugin that only includes the Android platform implementation. I don't want to implement the desktop version. I tried commenting out all the desktop macro code in lib.rs, but then I encountered an error when running cargo build.

MnlPhlp, 2025-02-12:
Could you show me the error please. This plugin only uses tauri for the android implementation and the js commands. Bluetooth on desktop and iOS works using the native implementation from btleplug

tinyvan, 2025-02-14:
thankyou
i am just developing a plugin too but i only finished android implementation,and the compilation failed due to lack of desktop implementation.and now i have known how to solve it by adding `#![cfg(mobile)]`

closed 2025-02-14 by tinyvan