MnlPhlp/tauri-plugin-blec #7: Permissions error when bluetooth scanning

timskillman, 2025-02-18:
Hi,

Trying to get your code working in Android but I can't seem to get permissions

**How I'm invoking the scan**

```
var channel = new __TAURI__.core.Channel();
channel.onmessage = function() {
    console.log("scanned");
}
__TAURI__.core.invoke('plugin:blec|scan', {
    timeout: 10000,
    onDevices: channel
}) 
```

**Startup Warning**
```
02-18 11:52:50.919 23927 24121 W cr_media: BLUETOOTH_CONNECT permission is missing.
02-18 11:52:50.929 23927 24121 W cr_media: getBluetoothAdapter() requires BLUETOOTH permission
02-18 11:52:50.929 23927 24121 W cr_media: registerBluetoothIntentsIfNeeded: Requires BLUETOOTH permission
```

**Scan Invoking Error**
```
02-18 11:54:09.903 24529 24529 W Activity: Can request only one set of permissions at a time
02-18 11:54:09.954 24529 24529 E Tauri/Console: File: http://tauri.localhost/bluetooth - Line 0 - Msg: Uncaught (in promise) Btleplug error: Runtime Error: Need android.permission.BLUETOOTH_SCAN permission for android.content.AttributionSource@791416b5: GattService registerScanner
```

Adding BLUETOOTH_SCAN or any of the other Android Bluetooth permissions to the AndroidManifest.xml makes no effect on the outcome.

Any help much appreciated!

Thanks

MnlPhlp, 2025-02-19:
Did you get a permission request from android?
If not check in the app settings if there is a denied permission that you can grant.

Maybe it is marked as denied if you tried it once without the permission in the manifest

MnlPhlp, 2025-02-19:
And make sure you are using the latest version of the crate `0.3.4`
You might need to do `cargo update` if there is an older version in your cargo.lock
`0.3.3` Fixed an error with permission requests

timskillman, 2025-02-20:
That worked!  Thank you for the update - much appreciated :-)

closed 2025-02-20 by timskillman