MnlPhlp/tauri-plugin-blec #8: macos failed

ks-zc, 2025-03-04:
blec.scan not allowed. Permissions associated with this command: blec:allow-scan, blec:default

MnlPhlp, 2025-03-05:
That means you are missing the permission `blec:default` in your capabilities file.
In the example this is in [src-tauri/capabilities/default.json](https://github.com/MnlPhlp/tauri-plugin-blec/blob/main/examples/plugin-blec-example/src-tauri/capabilities/default.json)

Looks like I missed explaining this in the readme. Will add it tomorrow

MnlPhlp, 2025-03-05:
These permissions are needed for all tauri commands. Explained [here](https://v2.tauri.app/security/capabilities/)

MnlPhlp, 2025-03-11:
I will close this, if you still have problems after setting the permissions feel free to reopen or create a new issue

closed 2025-03-11 by MnlPhlp