MnlPhlp/tauri-plugin-blec #11: Android don't get any reply

oddpxl, 2025-03-11:
On Android I can scan devices, see them and connect...

..but when I write to one with reply - I don't get a reply - regardless if I have connected or not.

The same code writes with reply and get responses on MacOS - same device - same code.

I can't seem to find anything strange in logcat either...

Any suggestions ?

oddpxl, 2025-03-12:
Here's two log examples...  MacOS / Android...

..please note the lack of replies on Android - ( despite the hardware device itself indicates being connected )

( sorry for it being verbose )

------ MAC OS -------

2025-03-12T01:09:38.695Z - -- START -- (Platform: macOS)
2025-03-12T01:09:38.696Z - Starting scan...
2025-03-12T01:09:38.697Z - Scanning status: active
2025-03-12T01:09:39.103Z - New batch of 1 devices
2025-03-12T01:09:41.611Z - Name: frodd - Address: fb5089c3-a62a-a014-0b37-3bf6e0a0ecda - Connected: false
2025-03-12T01:09:41.613Z -   Manufacturer(0xFFFF): [02 15 FD A5 06 93 A4 E2 4F B1 AF CF C6 EB 07 64 78 25 27 1B 4C B9 C9]
2025-03-12T01:09:41.615Z - ----> Found target device! <----
2025-03-12T01:09:41.618Z - UUID matched
2025-03-12T01:09:41.620Z - Device name matched
2025-03-12T01:09:41.622Z - Device details:
2025-03-12T01:09:41.624Z -   Address: fb5089c3-a62a-a014-0b37-3bf6e0a0ecda
2025-03-12T01:09:41.626Z -   Name: frodd
2025-03-12T01:09:41.628Z -   Services: []
2025-03-12T01:09:41.631Z - Target address: fb5089c3-a62a-a014-0b37-3bf6e0a0ecda
2025-03-12T01:09:41.633Z - --- STOPPING SCAN ---
2025-03-12T01:09:41.635Z - Scan stopped successfully
2025-03-12T01:09:41.637Z - Scan monitor aborted
2025-03-12T01:09:41.640Z - --- CONNECTING ---
2025-03-12T01:09:41.643Z - Attempting connection to: fb5089c3-a62a-a014-0b37-3bf6e0a0ecda
2025-03-12T01:09:42.799Z - Connection successful
2025-03-12T01:09:43.302Z - Subscribing to notifications on UUID: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
2025-03-12T01:09:43.337Z - Notification subscription successful
2025-03-12T01:09:43.841Z - Sending auth...
2025-03-12T01:09:43.954Z - Auth data sent successfully
2025-03-12T01:09:43.961Z - Sending temperature request...
2025-03-12T01:09:43.965Z - ============================================
2025-03-12T01:09:43.969Z - NOTIFICATION HANDLER CALLED
2025-03-12T01:09:43.973Z - TIMESTAMP: 1741741783
2025-03-12T01:09:43.977Z - ============================================
2025-03-12T01:09:43.980Z - Raw data length: 4
2025-03-12T01:09:43.983Z - Raw hex: [F3 00 F3 00]
2025-03-12T01:09:43.986Z - Protocol header (0xF3) detected
2025-03-12T01:09:43.990Z - Auth response: 00 -> accepted
2025-03-12T01:09:43.993Z - ============================================
2025-03-12T01:09:44.102Z - Temperature request sent successfully
2025-03-12T01:09:44.106Z - Sending humidity request...
2025-03-12T01:09:44.177Z - ============================================
2025-03-12T01:09:44.182Z - NOTIFICATION HANDLER CALLED
2025-03-12T01:09:44.185Z - TIMESTAMP: 1741741784
2025-03-12T01:09:44.189Z - ============================================
2025-03-12T01:09:44.193Z - Raw data length: 6
2025-03-12T01:09:44.196Z - Raw hex: [F3 10 F3 02 17 3D]
2025-03-12T01:09:44.200Z - Protocol header (0xF3) detected
2025-03-12T01:09:44.203Z - Temperature data: 17 3D -> 23.6°C
2025-03-12T01:09:44.207Z - ============================================
2025-03-12T01:09:44.327Z - Humidity request sent successfully
2025-03-12T01:09:44.332Z - Sending battery request...
2025-03-12T01:09:44.336Z - ============================================
2025-03-12T01:09:44.340Z - NOTIFICATION HANDLER CALLED
2025-03-12T01:09:44.344Z - TIMESTAMP: 1741741784
2025-03-12T01:09:44.347Z - ============================================
2025-03-12T01:09:44.351Z - Raw data length: 5
2025-03-12T01:09:44.355Z - Raw hex: [F3 12 F3 01 1F]
2025-03-12T01:09:44.358Z - Protocol header (0xF3) detected
2025-03-12T01:09:44.361Z - Humidity data: 1F -> 31%
2025-03-12T01:09:44.365Z - ============================================
2025-03-12T01:09:44.477Z - Battery request sent successfully
2025-03-12T01:09:44.481Z - Sending vibration request...
2025-03-12T01:09:44.485Z - ============================================
2025-03-12T01:09:44.489Z - NOTIFICATION HANDLER CALLED
2025-03-12T01:09:44.492Z - TIMESTAMP: 1741741784
2025-03-12T01:09:44.496Z - ============================================
[... 64 lines cut ...]
2025-03-12T01:06:22.546Z - Subscribing to notifications...
2025-03-12T01:06:22.550Z - Notification subscription successful
2025-03-12T01:06:22.552Z - Setting up Android notifications stream...
2025-03-12T01:06:22.555Z - Connected device: frodd
2025-03-12T01:06:22.557Z - Starting Android notification listener task
2025-03-12T01:06:23.545Z - Android notification setup complete
2025-03-12T01:06:24.542Z - Notification setup complete
2025-03-12T01:06:24.551Z - Sending auth...
2025-03-12T01:06:24.668Z - Auth data sent successfully
2025-03-12T01:06:24.671Z - Sending temperature request...
2025-03-12T01:06:24.820Z - Temperature request sent successfully
2025-03-12T01:06:24.827Z - Sending humidity request...
2025-03-12T01:06:24.972Z - Humidity request sent successfully
2025-03-12T01:06:24.976Z - Sending battery request...
2025-03-12T01:06:25.118Z - Battery request sent successfully
2025-03-12T01:06:25.152Z - Sending vibration request...
2025-03-12T01:06:25.268Z - Vibration request sent successfully
2025-03-12T01:06:35.270Z - Sending temperature request...
2025-03-12T01:06:35.396Z - Temperature request sent successfully
2025-03-12T01:06:35.429Z - Sending humidity request...
2025-03-12T01:06:35.542Z - Humidity request sent successfully
2025-03-12T01:06:35.576Z - Sending battery request...
2025-03-12T01:06:35.693Z - Battery request sent successfully
2025-03-12T01:06:35.723Z - Sending vibration request...
2025-03-12T01:06:35.842Z - Vibration request sent successfully

MnlPhlp, 2025-03-17:
Vy reply you mean a notification that is send by a subscribed characteristic?
I see the 'NOTIFICATION HANDLER CALLED' in your log.

So does this mean on android the notification handler is not correctly called?

oddpxl, 2025-03-17:
Exactly.

I even forked the project to add debug messages - just to investigate - and the callback in subscribe never get called on Android - and it does on MacOS ( using the same code and bluetooth device )

The bluetooth device clearly receives all messages ( the hardware tells me it connected and received authentication ) and I receive no errors from BLEC.

MnlPhlp, 2025-03-20:
I just hat the same problem.
There was en error with how i returned the values from Android.
Please try again with the newest version (0.4.5)

oddpxl, 2025-03-20:
Works now !!

My device return values as it should. Great stuff. Thanks.

closed 2025-03-20 by oddpxl

GaZaTu, 2025-03-26:
I have this exact same problem even with the latest version of this plugin.
I can connect and sendString, but i don't receive anything in my subscribe handler even tho it works on my linux pc.
Is there some special magic i need to enable this functionality?