MnlPhlp/tauri-plugin-blec #14: requestMtu for large transmissions

GaZaTu, 2025-03-31:
call https://developer.android.com/reference/android/bluetooth/BluetoothGatt#requestMtu(int) before the following please or add a new rust/js api to do it manually.

https://github.com/MnlPhlp/tauri-plugin-blec/blob/ff0ccb8cf3c7baa2414d50bd043dac35d15cba8c/android/src/main/java/Peripheral.kt#L166

currently bt notifications on android are capped to 20 bytes of data because no MTU is being set.

MnlPhlp, 2025-04-05:
Btleplug has an open PR to add mtu requests [here](https://github.com/deviceplug/btleplug/pull/407)

For now I added a request for a higher MTU on android, but there is no way to return the actual MTU because i use the btleplug structs internally. Usually it should be set to 517. This is the same that newer Android versions set [automatically](https://developer.android.com/reference/android/bluetooth/BluetoothGatt#requestMtu(int))
Please try with the newest Version (0.5.3)

GaZaTu, 2025-04-08:
idk, i forgot that i'm using the rust side so this change doesn't even affect me xd.
but this isn't an issue for me anyway now since i managed to force a higher mtu on the server side.

MnlPhlp, 2025-04-08:
This does affect the rust side.
It is implemented in the part calling the android plugin, which is used by the rust backend as well as the js commands.

MnlPhlp, 2025-04-08:
So can I close this?

GaZaTu, 2025-04-08:
sure

closed 2025-04-08 by MnlPhlp