MnlPhlp/tauri-plugin-blec #15: Android BLE: Promise hangs after sending data to thermal printer using send method

git0x4c, 2025-04-24:
📋 Steps to Reproduce:
1. Perform a normal scan → connect → send data → ✅ success

2. Disconnect → reconnect → send data → ✅ success

3. Send data again → ❌ the promise returned by send hangs indefinitely (no resolution, no rejection)

🐞 Issue:
After step 3, the promise returned by the send method never resolves or rejects.

No error is thrown or caught.

Any subsequent calls to methods from @mnlphlp/plugin-blec also hang and do not proceed.

This seems to affect Android specifically.

💡 Expected Behavior:
The send promise should either resolve (if successful) or reject (if failed), allowing further logic to continue.

📱 Environment:
Platform: Android
Library: 
"@mnlphlp/plugin-blec": "^0.4.1"
tauri-plugin-blec = "0.4"
tokio = { version = "1.44.2", features = ["time"] }
BLE Device: Thermal Printer

1. rustc 1.86.0 (05f9846f8 2025-03-31)
2. cargo 1.86.0 (adf9b6ad1 2025-02-28)
3. node v18.19.0
4. android 15 & android 14

- 伪代码

```ts
import {
	connect,
	startScan,
	stopScan,
	BleDevice,
	getConnectionUpdates,
	getScanningUpdates,
	disconnect, send, subscribe, unsubscribe,
} from '@mnlphlp/plugin-blec'
import {invoke} from "@tauri-apps/api/core";

// Get characteristic values ​​and send data to thermal printer
const handleGetDeviceInfo = async () => {
	let characteristicId = ''
	try {
		const services = currentDevice?.services || []

		const res: ServiceItem[] = await invoke('blec_discover_services', {
			address: currentDevice?.address
		})

		console.log('res', res)
		// Finding characteristic values ​​for a service
		services.forEach(serviceId => {
			const id = getCharacteristicIdForService(res, serviceId)
			if (id) {
				characteristicId = id
				return false
			}
		})

		if (!characteristicId) {
			console.warn('not found characteristicId')
			return
		}

		const tspl = [
			'',
			'SIZE 60 mm,40 mm',
			'GAP 2 mm,0',
			'CLS',
			'QRCODE 100,100,L,5,A,0,"https://google.com"',
			'TEXT 100,220,"TSS24.BF2",0,1,1,"google.com"',
			'PRINT 1,1'
		].join('\n')
		const buffer = strToUTF8ArrayBuffer(tspl)

		console.log('sendData', {
			characteristicId,
			data: buffer.buffer,
		})
		setTimeout(async () => {
			console.log('subscribe', characteristicId)
			try {
				await subscribe(characteristicId, (data) => {
					console.log('subscribe data', data)
				})
			} catch (e) {
				console.error('Error subscribing:', e)
			} finally {
				console.log('finally subscribe', characteristicId)
				if (characteristicId) {
					await unsubscribe(characteristicId)
				}
			}
		}, 2000)
		await send(characteristicId, buffer.buffer as Uint8Array, 'withoutResponse')
		console.log('send data -> done')
	} catch (e) {
		console.error('Error sending data:', e)
	} finally {
		console.log('finally', characteristicId)
		// if (characteristicId) {
		// 	await unsubscribe(characteristicId)
		// }
	}
}
```
```react
const handleDisconnect = useCallback(async () => {
	try {
		console.log('handleDisconnect disconnect....')
		await disconnect() // <- The promise is suspended
		console.log('Disconnected')
		setConnected(false)
		setCurrentDevice(null)
	} catch (e) {
		console.error('Error disconnecting:', e)
	}
}, [currentDevice])
```

```rust
// Get the characteristic value blec_discover_services
use tauri_plugin_blec::models::Service;
use tokio;
#[tauri::command]
async fn blec_discover_services(address: String) -> Result<Vec<Service>, String> {

    let handler = tauri_plugin_blec::get_handler()
        .map_err(|e| format!("BLE handler not available: {:?}", e))?;

    match tokio::time::timeout(
        std::time::Duration::from_secs(10), // 10s timeout
        handler.discover_services(&address)
    ).await {
        Ok(result) => result.map_err(|e| format!("Service discovery failed: {:?}", e)),
        Err(_) => Err("Service discovery timed out after 10 seconds".to_string()),
    }
}
```

[image]

[image]

git0x4c, 2025-04-24:
✅ Temporary Workaround:
After reconnecting to the device, manually clear the current BLE device list and restart the scan before attempting to send data again.
This resolves the issue where the send promise gets stuck.

This issue can also be resolved by restarting the app, which suggests that some internal BLE state or resource is not properly reset after reconnecting.

hungnq139, 2025-12-23:
I encountered a similar case. After tracing the flow with RUST_LOG=trace, I found the error:
+ It turned out that my service configured the characteristic with only 8 bytes, while I was sending more than that.
+ The library automatically chunks the data, but does not support handling it.
+ An error occurs, but the library does not handle it, causing the code to hang at the await function.
+ Because it hangs, all subsequent commands are either queued or fail.

→ I simply fixed it by correcting the characteristic configuration on the service side.

MnlPhlp, 2026-10-29:
A lot changed in the send logic since your initial issue. Is this still a problem or can this issue be closed?