MnlPhlp/tauri-plugin-blec #17: BLE devices can be discovered but fail to connect on Huawei devices

TimKeeper, 2025-05-12:
**Description:**
I encountered an issue where BLE devices can be successfully scanned and discovered, but the connection attempt always fails. This issue seems to occur specifically on Huawei devices.

**Steps to Reproduce:**
1. Use a Huawei device (e.g., Huawei P60 or HUAWEI Pura 70 Ultra).
2. Scan for nearby BLE devices.
3. Attempt to connect to a discovered BLE device.

**Expected Behavior:**
The BLE device should connect successfully after being discovered.

**Tested Devices:**
Huawei P60
HUAWEI Pura 70 Ultra

MnlPhlp, 2025-05-12:
Do you have any logs from those failed connections? I don't own a Huawei device so I can't try It myself.

TimKeeper, 2025-05-12:
It can be confirmed that the BLE address scanned by other Android devices is in UUID format, while on Huawei devices, the scanned address is in MAC address format. Additionally, the Bluetooth connection can be successfully established on Huawei devices, but there are no results when subscribing to messages.

In my code, the message subscription is also using the UUID format. I'm not sure if this discrepancy between the MAC address and UUID is causing the issue. Could this be the reason for the subscription failure?

[image]

TimKeeper, 2025-05-15:
To supplement the issue, after debugging, I found that the device can connect successfully, and data subscription also works (the subscribeString method executes successfully, although I'm not entirely sure if it's actually functioning as expected). The server can receive messages from the client, but messages sent from the server to the client cannot be received through the subscription.

MnlPhlp, 2025-05-15:
There might be an issue with the plugin expecting a UUID address on Android. Can you get any errors from logcat or the Terminal when running the tauri dev command?

TimKeeper, 2025-05-16:
I have tested using tauri dev, tauri ios dev, and tauri android dev, and there are no error messages in any of these environments. Additionally, everything works fine on macOS, Windows, iOS, and other Android devices. The issue only occurs on Huawei devices, so I suspect it might be related to the HarmonyOS system. However, I currently have no way to confirm if this is the root cause.

Soulkau, 2025-12-26:
The subscription issue should be fixed in (#44). As far as I know Huawei has custom android based builds, perhaps they are using deprecated android api even if os build based on newer android version. I had same subscription problem with my android 12 phone without the fix.