MnlPhlp/tauri-plugin-blec #18: connect() error handling

arashi-dev, 2025-05-15:
How can I handle connection errors when calling connect()?

as I checked the source code, it's wrapped in try catch, which prevents me to add my own try catch and handle fail/success events. is there any reason you have added them? It seems that it was just for a development testing but committed accidentally.
```ts
export async function connect(address: string, onDisconnect: (() => void) | null) {
  console.log('connect', address)
  let disconnectChannel = new Channel()
  if (onDisconnect) {
    disconnectChannel.onmessage = onDisconnect
  }
  try {
    await invoke('plugin:blec|connect', {
      address: address,
      onDisconnect: disconnectChannel
    })
  } catch (e) {
    console.error(e)
  }
}

```

MnlPhlp, 2025-05-16:
can this be closed with your PR merged?

MnlPhlp, 2025-05-16:
I will close this because your PR fixed the issue

closed 2025-05-16 by MnlPhlp

mike-lloyd03, 2025-07-31:
@MnlPhlp I'm running into this issue on 0.4.1 which appears to have been released before #19 was merged. Can you publish a new version?

MnlPhlp, 2025-07-31:
@mike-lloyd03 Thank your for the info. I totally forgot to release the npm package. I published a new version (0.5.3 to match the crate)