MnlPhlp/tauri-plugin-blec #29: Scan required before connect?

mike-lloyd03, 2025-08-21:
I have some BLE devices stored in my mobile app's local DB. They are nearby and advertising. When I try to connect to them, I usually get the `There is no peripheral with id: xx:xx:xx:xx:xx` error.

However, if I scan for devices first, I'm able to connect to this device. Is this the expected behavior?

If so, what is the idomatic way to handle connecting to a new, known device?

```typescript
export async function connectDevice(device: Device) {
	if (device.bt_address) {
		try {
			startScan(() => {}, 4000);
			await connect(device.bt_address, () => {
				if (appState.connectedDevice?.bt_address == device.bt_address) {
					appState.connectedDevice = null;
				}
			});
			appState.connectedDevice = device;
			info(`Connected to: ${device.name ?? device.bt_address}`);
		} catch (e) {
			throw e;
		}
	}
}
```

MnlPhlp, 2025-09-09:
The problem here is that the underlying btleplug library keeps a platform specific list of the actual device handles that are needed to connect. This is created when scanning.

I think the best way to connect to a known device would be to start a scan and then connect once the device is discovered.

Your function probably does not work because the startScan return immediately and the connect happens without any time for the scan. The timeout given to scan just specifies how long the scan runs and triggers the callback, the scan function itself always returns directly.

mike-lloyd03, 2025-09-15:
Alright I implemented this. Seems like it might work.

If I await startScan, will it block until the scan is complete?